MatrixPilot/MatrixPilot #14: DESIRED_SPEED reported as not working

phollands, 2016-04-30:
Guilio reported that DESIRED_SPEED in options.h was not working, and that he could not change the desired speed from within a Logo flight plan.

Looking at this just now I see that DESIRED_SPEED appears to only be set from options.h [in altitudeCntrl.c ](https://github.com/MatrixPilot/MatrixPilot/blob/master/MatrixPilot/altitudeCntrl.c#L155) if USE_MAVLINK == 1. And if Guilio was using the increment and decrement speed instructions from Logo,
then he will have been incrementing and decrementing the variable desiredSpeed from a starting point of 0.

I am not sure why the setting of DESIRED_SPEED is linked to using MAVLink ?

[image]

As a result of a recent commit,  mavlink is enabled at the moment with SERIAL_OUTPUT_FORMAT  defined as SERIAL_MAVLINK and not with USE_MAVLINK.

KeesGuijt, 2016-05-02:
The #else (third line in the screenshot) makes this code apply only when SPEED_CONTROL == 0, so this is not a problem. 

DESIRED_SPEED should now be read (once) by config.c.
 (config.c) altit.DesiredSpeed = ini_getf(strAltitude, "desired_speed", DESIRED_SPEED, strConfigFile);
 (altitudeCtrl.c) desiredSpeed = altit.DesiredSpeed * 10;

I've monitored desiredSpeed while running LOGO. It changes when needed. However no effect on the airspeed.

I hope to test some more with this script:

	SET_ALT(100)

	// Go Home and point North
	HOME

	REPEAT_FOREVER
		DO_ARG(SQUARE, 400)
	END

	TO (SQUARE)
                SET_SPEED(9)
		REPEAT(4)
			FD_PARAM
			RT(90)
			SPEED_INCREASE(1)
		END
	END

Best regards,
Kees

phollands, 2016-05-15:
Kees is discussing this issue more widely [on uavdevboard at this location.](https://groups.google.com/d/msg/uavdevboard/zjkci1hluSU/Z8mqtJVOBgAJ)

phollands, 2016-08-13:
Kees, can you confirm that DESIRED SPEED is now working correctly ?

KeesGuijt, 2016-08-13:
Pete,

It works, in a way, for gliders only. But I hope it will be replaced with [airspeed modulation of gains](https://groups.google.com/forum/#!msg/uavdevboard/zjkci1hluSU/tUxDyBw_CwAJ) in the future.

Kees

phollands, 2017-03-02:
It is my understanding that Bill hopes to work on Speed Control with the motor working, laster on in the Summer.