MatrixPilot/MatrixPilot #17: SUE over MAVLINK needs to be updated

giukio, 2016-05-24:
Hello,

I've recently switched from SERIAL_UDB_EXTRA to SERIAL_MAVLINK and I think that SUE sent over MAV_DATA_STREAM_EXTRA1 needs to be updated to the version currently in telemetry.c.

In fact, flan exits with an error while parsing the telemetry file:
Exception in Tkinter callback
[...]
AttributeError: flight_log_book instance has no attribute 'elevator_output_channel'

By comparing a F2 string I notice:

1. SUE over MAVLINK (MavSUE from now on) still has the bmv field after the cpu entry
MavSUE: cpu17:	bmv0:	as1152:
SUE:	cpu17:			as902:

2. MavSUE sends 10 input and output channels while SUE only those defined in options.h
MavSUE: p1i2910:p2i2975:p3i2233:p4i2997:p5i2997:p6i1977:p7i0:p8i0:p9i0:p10i0:
SUE:	p1i2967:p2i3047:p3i3050:p4i2997:p5i4018:p6i2997:
MavSUE: p1o2869:p2o2919:p3o2279:p4o2996:p5o0:p6o0:p7o0:p8o0:p9o0:p10o0:
SUE:	p1o2823:p2o2972:p3o2622:p4o2997:

3. MavSUE does not include the lex ley lez fields:
MavSUE: imz11:						fgs5C0:
SUE:	imz92:		lex0:ley0:lez-1:		fgs9E0:

4. MavSUE has a stk field (not sure what that is), while SUE doesn't

5. MavSUE does not include the AF and battery data fields
SUE: 	AF409,40,-1966:	bmv110:mA23:mAh882:

Best Regards,
Giulio

phollands, 2016-05-24:
Giulio, Thanks for noting all the specific issues. Yes, after all the changes in HelicalTurns that are now in Master, plus the later changes in for battery consumption, SUE over MAVLink definitely needs updating. It is quite a big job, but I guess it falls to me.  In general I did not want to do any more updates to ascii SUE (from about 2 years ago), but I did this special for Bill who does not yet use MAVLink. But this will be the last time that I update SUE Ascii, and in the future MatrixPilot MAVLink packets will prevail as the place of innovation. MAVLink wins on nearly every aspect, except when first debugging a serial link. 

It will probably be quite a few weeks, and may be a couple of months, before I can get around to coding this up. The first step will be to ensure that the MAVLink generation suite in Master is working correctly. Then after that is proven, I will move onto fixing up MAVSue.

If you want to move faster, then I can give you guidance, and you can see how you get on. 

Best wishes, Pete

giukio, 2016-05-24:
If you can give me guidance, I could try to work on it! I don't guarantee to be faster than you, but mavlink interests me as I would also like to add a few messages (like wind direction, if not already present).

I'm just worried about flan, I don't know how to program in pyton and I hope it shouldn't be modified.

Best regards,
Giulio

phollands, 2016-05-26:
I don't think you can escape learning some Python. It is a interpretive language which helps with experimentation and ease of learning. (Does not need compiling).  But I will certainly help out with Flan.pyw (after all I wrote it, and it's my mess of a learning journey for when I learn't python). The other Ground Control Station of interest is MAVProxy from Tridge, which is also written in Python (and my better in terms of code). Python will let you easily program the use of any message. Although the object oriented aspect of python does mean that mavproxy will graph any new message without any further programming (it will see the message using the generated python from the XML specifications).

Your first goal should be to re-check whether the mavlink generator  (written in Python) is still working for MatrixPilot. You will find it at[ this location in the distribution.](https://github.com/phollands/MatrixPilot/blob/master/Tools/MAVLink/mavlink/pymavlink/generator/gen_MatrixPilot.py) You will need to install python 2.7.x and then run the program gen_MatrixPilot.py to see what happens. If it generates the code correctly, and updates all the *.h files for MAVlink then that is good. 

There is a[ test system](https://github.com/phollands/MatrixPilot/blob/master/MatrixPilot/MAVLink.c#L85) for checking that all the .h files work correctly. Optionally I could tell you how to enable that in MatrixPilot to run a test. It will run several hundred tests. There are some [more details in the Wiki.](https://github.com/MatrixPilot/MatrixPilot/wiki/MavlinkGeneratingHeaders).

If that is all OK. then you can start to edit new [MatrixPilot XML messages for SUE MAVLink](https://github.com/MatrixPilot/MatrixPilot/blob/master/Tools/MAVLink/mavlink/message_definitions/v1.0/matrixpilot.xml).  You then run the MatrixPilot generator again, and then all the .h C include files will be generated AND all the python code for MAVproxy use, and flan.pyw use (for parsing the data). Note, if you answer "yes" to the various questions, then the .h files in the main MatrixPilot code base of your repository will be update (but there will no git commit, just working copy updated).

It it then a question of sending the correct new MAVLink messages for SUE from MatrixPilot. I would then help with using them in flan.pyw to re-generate the original SUE Ascii files. I will need to also update the flan.pyw parsing of Mavlink.

That is the about it.

Best wishes, Pete

excalbian, 2016-09-25:
Giulio / Pete, any recent progress on this? I got a wild herring when my LHS had the new Spektrum AR6600T six channel receiver with telemetry to try connecting my UDB through the X-Bus (https://www.spektrumrc.com/ProdInfo/Files/SPM_Telemetry_Developers_Specs.pdf). I found an old thread from Ric back in February with the same idea, but it doesn't look like anyone has done much on this yet.

Anyways, I figured before trying that, I'd need a real time telemetry option that was proven (so I could compare with what I got through the Spectrum) which I think means I need to jump into MAV-link (as a longtime FLAN / SUE user). I then saw this issue and figured it might help me as well to have both the ground station (MAV-link) and SUE data packets when debugging things.

phollands, 2016-09-25:
Adam, I started looking at this issue last week, and expect to make progress next week, especially if the weather is non flyable :-). It is quite a lot of work (more than what I wrote above)  and may take a few weeks to complete.

KeesGuijt, 2016-10-13:
Hi Pete,

Nice to see progress here, great.
I've just finished my hilsim/mavlink/qgc setup for testing. I've found some issues and wishes with onboard parameters. When would be a good time to report those?

Regards,
Kees

phollands, 2016-10-13:
Hi Kees,

Please can you open a new issue for each problem that you find. Yes, now is a good time, as I have my head around mavlink related issues and I'm re-familiarising myself with the code.

 Yes PR #70 mavlink_core_update has the new foundations using the up to date mavlink generators. The PR is intended to be just for the core foundation (e.g. include files) of mavlink and no more. 

 I had some problems with the python mavlink library code having changed, so that flan.pyw no longer worked with our mavlink logs captured in the openlog (they have no time stamps). But that is fixed now in the new commits that I've made.

Best wishes, Pete

phollands, 2016-10-20:
Giulio,

Fixed in Pull Request #90.   I hope the voltage and current measurement all come through correctly now, when using mavlink.

Best wishes, Pete

closed 2016-10-20 by phollands

giukio, 2016-10-25:
Hi,

I'm currently really busy as I was on this summer (my dissertation day is
getting closer!).
Unfortunately I'm not able to test that soon but I'm trying to keep up
reading all the emails.

Just wanted to remind that if we want to fully support the mavlink osd
extra we need to send also the wind speed message that ardupilot uses (I
just don't remember it now).

Going a little bit off topic, speaking about OSDs I suggest you to take a
look at alce osd, it seems really cool to me, and plus uses a dspic (if we
ever need to tweak its firmware for whatever reason).

I can't wait to start flying again and test out all the new code!

Best wishes,
Giulio

phollands, 2016-10-26:
HI Giulio,

I would like to encourage you in your dissertation. It will be a marathon, and I hope you can just keep on going and going to the end.  I have every confidence in your abilities, although I'm sure you are being well tested in your academic skills.  Good Luck, and I look forward to hearing from you later in the year, after your work is completed. 

Best wishes, Pete