MatrixPilot/MatrixPilot #33: WAYPOINT_RADIUS is redefined

elgarbe, 2016-08-13:
I have this warning when I compile master brach

```
In file included from ../../MatrixPilot/flightplan-waypoints.c:44:0:
../../MatrixPilot/waypoints.h:26:0: warning: "WAYPOINT_RADIUS" redefined
../../MatrixPilot/flightplan.h:28:0: note: this is the location of the previous definition
```

KeesGuijt, 2016-08-16:
Hi,

WAYPOINT_RADIUS should be defined only in flightplan.h, and removed from both
waypoints.h and flightplan-logo.h.

Regards,
Kees

phollands, 2016-08-16:
Kees is right.

But there  is problem with some of the waypoints.h file lines being moved
to flightplan.h . It breaks flan.pyw which is looking for these lines in
waypoints.h . It only breaks flan.pyw when it is trying to plot the
waypoints in Google Earth KMZ files.

That is why I have not got around to looking at this issue yet myself.

Either flan.pyw has to change, or we need to revert some key lines back
into waypoints.h . However, Robert has stated that he moved them to
flightplan.h because he wants to run both Logo and Waypoint navigation
methods in the same flight (or atleast in the same binary). I'm not sure
why that is a useful use case ?

Best wishes, Pete

phollands, 2016-09-16:
If one defines all the options in waypoints.h that could be correctly put there (as they were originally in waypoints.h ), then there are four errors:-

> In file included from ../../MatrixPilot/flightplan-waypoints.c:44:0:
> ../../MatrixPilot/waypoints.h:26:0: warning: "WAYPOINT_RADIUS" redefined
> ../../MatrixPilot/flightplan.h:28:0: note: this is the location of the previous definition
> ../../MatrixPilot/waypoints.h:28:0: warning: "CAM_VIEW_LAUNCH" redefined
> ../../MatrixPilot/flightplan.h:30:0: note: this is the location of the previous definition
> ../../MatrixPilot/waypoints.h:50:0: warning: "USE_FIXED_ORIGIN" redefined
> ../../MatrixPilot/flightplan.h:52:0: note: this is the location of the previous definition
> ../../MatrixPilot/waypoints.h:51:0: warning: "FIXED_ORIGIN_LOCATION" redefined
> ../../MatrixPilot/flightplan.h:56:0: note: this is the location of the previous definition

I'm not much of a fan of the entire idea of running the flightplan-logo.h alongside waypoints.h, but if we have to do this, then I think I will give unique names to the parameters in Logo, and leave the old waypoints.h names as they are. e.g. LOGO_WAYPOINT_RADIUS and a similar convention for the other 3 parameters. But it is going to be complex in the navigation code, if it needs to decide always whether it is running LOGO or waypoints before deciding whether to use a fixed origin or relative to bootup. 

I have to say, that I would rather we just returned to having two separate flight plan systems, and that they never run together at the same time. i.e. Revert to what we had originally. I just don't see the use case for running both systems in the autopilot at the same time.

This issue is now getting in the way of releasing a new version of MatrixPilot (5.0).

Can we agree to keep the flight planning systems separate for now ? In MP5.0 ?

Inspirati, 2016-09-19:
The goal was to have support for both compiled in, and then have a runtime
option to select which flightplan method is used on any given flight. Like
much of my work on adding dynamic parametisation - being a move towards
making MatrixPilot more run-time configurable without needing the
rebuild/reflash cycle.

phollands, 2016-09-19:
Hi Robert, thanks for your comment. In the current fix in the pull request, I have essentially kept your code,  so both Logo and waypoint flight planning systems are still both compiled into the firmware, ready for run-time configurable options.

PR #53 (2016-09-19): Move common parameters for flight plans in  to options.h.
In current master, WAYPOINT_RADIUS, USE_FIXED_ORIGIN, and FIXED_ORIGIN_LOCATION are common to both the Logo and Waypoints.h flight plan types. The code compiles both flight plan types at all times nowadays. (A move made by Robert who I believe had a vision parameters to be dynamically configurable without a re-compile). Because all code is compiled, these parameters can no longer be held in both flightplans-logo.h and wayoints.h or they will cause a warning from the compiler about parameters being re-defined.

This commit simply moves t [... cut]
closed 2016-09-19 by phollands