MatrixPilot/MatrixPilot #54: Consistent naming of options files in the Config directory.

phollands, 2016-09-19:
Now that most of the options files are all in the Config directory, it is readily apparent that we have a variety of naming conventions which looks unsightly. See the screenshot below ...

[image]

With an upcoming release of of MatrixPilot 5.0 in mind, I am intending to change the files as shown below. Any comments about this are welcome.

I intend to rename the files as:-

```
flightplan-logo.h        stays as           flightplan-logo.h
waypoints.h              becomes            flightplan-waypoints.h
airspeed_options.h       becomes            options_airspeed.h
mavlink_options.h        becomes            options_mavlink.h
nv_memory_options.h      becomes            options_nv_memory.h
options_auav3.h          stays as           options_auav3.h
manetometerOptions.h     becomes             @options_magnetometer.h
osd_config.h             becomes            options_osd.h
ports_config.h           becomes            options_ports.h
ServoMix_options.h       becomes            options_servo_miix.h

```

Best wishes, Pete

chiloschista, 2016-09-19:
> config_directory_mplab-x

Pete,
I like the idea of a more uniform way to call the various options.
Would also like to have only one options.h appearing in the projects 
tab, as I still see options.h appearing three times. I usually use the 
first one, but can't understand why the other two are there. Maybe from 
Cessna and CloudsFly folder? There are several other files doubled. I 
use MPLAB IDE.

Best regards,
Ric

---
Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

phollands, 2016-09-19:
Hi Ric, 

Thanks for your comments and the timely reminder about support for the MPLAB IDE. 

Personally, I want to drop support for the MPLAB IDE and have everyone move to MPLAB-X for MatrixPilot 5.0.  :-)  There is overwhelming logic for the move, and for dropping MPLAB IDE support. We will see how that goes down with everyone.

I do recommend moving to the use of MPLAB-X.

I will reflect on whether I want to do all the extra work for MPLAB IDE. Just to put this in perspective: there are currently 3 main projects (MatrixPilot, RollPitchYaw, and LedTest) and 3 main boards (UDB4, UDB5 and AUAV3), which when combined together makes 9 project configurations to setup in the MPLAB-X IDE, and if I do this in MPLAB IDE, that becomes another 9, which is a total of 18 project builds to test. 9 is bad enough, and 18 is just too many. Besides which MPLAB-X integrates with Git out of the box, has many additional and useful features, and is fully supported by Microchip, and is now on version 3.4.  MPLAB IDE is now a legacy environment with no new patches or releases .

Best wishes, Pete

KeesGuijt, 2016-09-19:
Pete,

The new names look nice.

Kees

KeesGuijt, 2016-09-20:
Pete,
Make that options_servo_mix.h

and what about splitting gain_variables.h and placing 
#define GAINS_VARIABLE                      0
#define ALTITUDE_GAINS_VARIABLE             0
in
options_gains_variables.h?

Best wishes, Kees

phollands, 2016-09-20:
I'll take a look at making options_gains_variables.h and see how hard it is
to split it out.

Best wishes, Pete

phollands, 2016-09-20:
This issue will be fixed by this pull request #56

phollands, 2016-09-21:
Fixed in Pull Requeste #56

closed 2016-09-21 by phollands