MatrixPilot/MatrixPilot #107: Estimated Airspeed Fluctuating

phollands, 2016-12-16:
Myself and Kees have noticed that the estimated air speed is fluctuating a  during flights. As an aircraft should be controlling it's desired speed against the actual air speed, this fluctuation should be understood and minimised. It is possible to put a filter on the airspeed, but the time has come, when we should understand in detail why the estimation is fluctuating.

For an example of this happening, look at the left hand side of the video which shows airspeed against ground speed in the On Screen Display analysis for one of[ Leo's later flights.](https://youtu.be/jIWJIbq714Q?t=359). Leo's flight is useful because he has long segments of straight level flight.

giukio, 2016-12-17:
I agree with you, I've also noticed this earlier this year. You can see in
in my videos as well.

I wouldn't put a filter on the airspeed though, as it would only mask the
effects (without solving the cause) by limiting the bandwidth of the
signal. If we want to control airspeed (count me in) that is not the
optimal choice in my opinion.

I tried to investigate that back in  April, I have a vague memory that was
related to the accelerometers integration, but don't count on this hint, I
could be totally wrong!

Best wishes,
Giulio

phollands, 2016-12-17:
The graph below shows IMU3DSpeed against Estimated Speed and Speed Over the Ground.

 The graph confirms that the source of the airspeed fluctuations is from the estimation of the IMU Ground speed. (IMU3dSpeed).

 The IMU3Dspeed is calculated in libre office spreadsheet from the following type of formula
=INT(SQRT((BH2^2+BI2^2+BJ2^2))+0.5)
The the relevant columns in the spreadsheet are IMUVelocityX (BH), IMUVelocityY (BI), IMUVelocityZ (BJ).

[image]

The data in the above graph is from the end of a log file for Flight 00028 (Log00455) of my White Tip Alpha flying a UDB5, Ublox at 2Hz, and using latest master as of 26th October 2016. 

The graph of the data, after the plane has landed, and the SOG is close to zero, is particularly useful and interesting.

I hope to dig into the data and calculations behind IMU3DSpeed next week.

chiloschista, 2016-12-17:
Hi Pete,
are you sure about the accuracy of SOG? I have already noticed important errors.
Should be checked with a true airspeed  sensor⁣.
Best regards,
Ric

Inviato da TypeApp ​

phollands, 2016-12-17:
Thanks Ric, I will keep issue of the accuracy of SOG for later. For now, I want to just deal with the noise that is clearly shown on the airspeed (blue line above), IMU3DSpeed (yellow) in relation to the steadier line of the Speed Over the Ground (SOG2) which is in red.

phollands, 2016-12-17:
If anyone understand the purpose of the following code in deadreckoning.c , then I would be grateful for your comments:-

```
IMUvelocityx.WW = IMUintegralAccelerationx.WW +
			                  __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[0]);
IMUvelocityy.WW = IMUintegralAccelerationy.WW +
			                  __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[1]);
IMUvelocityz.WW = IMUintegralAccelerationz.WW +
			                  __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[2]);
```
If i graph the IMU3dSpeed, SOG2, and also the magnitude of locationErrorEarth[] vector, then I do get some correlations between the noise on IMU3DSpeed and the size of locationErrorEarth. I shall look at this more next week.  Best wishes, Pete

[image]

LE_MAG_LRGE is Location Error Magnitude Large. The Large refers to scaling the LocationErrorEarth magnitude by 50 so that the data can be seen on the same scale as the Speeds on the same graph.
I note that ONE_OVER_TAU is effectively 2.5. And that 2.5*100 is 40. So the plot on my graph is close in size to the term in the above code (50 versus 40 in scaling factor).

phollands, 2016-12-18:
In the enclosed graph, I have used the telemetry to subtract out the velocity components that were added in associated with location_error_earth. The result is a large reduction in noise in the magnitude of the IMU velocity vector.

Here is graph of the same telemetry data as used previously in above comments. This time, IMU_V_MAG_adj is IMUVelocity magnitude calculated after subtracting each of the velocity components attributable to location_error_earth. I'll explain that more in a moment, but for now, here is the graph ....

[image]

Much of the noise that was on the blue line (IMU_V_MAG) [Magnitude of the IMU velocity vector[  has gone on the red line (IMU_V_MAG_adj  [Magnitude of the IMU velocity vector adjusted to subtract out the terms put in by dead reckoning for location_error_earth].

The exact calculations is to take each axis vector in turn, and subtract the terms that were added by[ this code:-](https://github.com/phollands/MatrixPilot/blob/535831c5a21c23ab542f44c345b33f1a9ef5a55f/libDCM/deadReckoning.c#L121)
```
IMUvelocityx.WW = IMUintegralAccelerationx.WW +
			                  __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[0]);
IMUvelocityy.WW = IMUintegralAccelerationy.WW +
			                  __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[1]);
IMUvelocityz.WW = IMUintegralAccelerationz.WW +
			                  __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[2]);
```
For example, The inverse operation of that code is simply to multiply locationErrorEarth[0] by 40 and subtract that from IMUvelocityx.WW. The value 40 is derived from TAU being 2.5. So ONE_OVER_TAU is effectively 1/2.5 which wen multiplied by 100 (see code above) is 40.  I do the same calculation for IMUvelocity.y IMUvelocity.z and then take the magnitude of the total velocity vector and call it IMU_V_MAG_adj.  All of that is done in the telemetry csv spreadsheet post flight.

The improvement is substantial but I don't know the purpose of the code above that uses locationErrorEarth. That has still to be understood. Here is a screenshot of the entire graph.

[image]

phollands, 2016-12-18:
A more detailed zoom in on the same graph, to show the reduction of noise in more detail ...

[image]

phollands, 2016-12-19:
My current hypothesis is that the airspeed fluctuations, are caused by the location_error_earth[x], [y], [z] terms being rather large as well as variable (changing). They are typically of the order of 3-5 meters, and are sometimes 10 meters, occasionally 15 meters.  A change of 5 meters in location_earth_error_earth is going to immediately change the air velocity by 5*40 centimeters / second which is 200 centimeters / second or 2.0 meters / second. which is the sort of order of change that we have been seeing in the telemetry. 

[image]

One aspect of location_error_earth, is that is it calculated only on integer variables that represent location as the number of meters (not centimeters). So  then applying that location_error_earth to calculate an additional error in ground speed that is measured in centimeters / second is  going to introduce some noise from the rounded integers of location_error_earth . I will look at the noise in location_error_earth in more detail shortly.

phollands, 2016-12-20:
An update for those of you following this issue: One question that I have needed to answer is "If location_error_earth can be as much as 5 to 10 meters, why do I not see that in the IMU track in Google Earth when compared to the GPS track in Google Earth ?  I think the answer is that DR_TAU (Dead Reckoning TAU) is set to 2.5 seconds. i.e. the firmware expects to remove any IMU location error over the next two seconds. And also, on my flight LOG00455 shown above, the Ublox is updating twice per second. So location error earth (which has noise on it) is only applied by about 1/2.5 * 1/2 on my flights which is 1/5th, between any two  updates from the GPS.  i.e. a 5 meter location_error_earth, will only have 1 meter applied to the IMUlocation before the next GPS update arrives. So this probably has a large smoothing effect of any noise in the GPS location when applied to IMUlocation. But IMU velocity, calculated in centimeters / second, is more sensitive it's calculations to the noise in location_error_earth.
I think that a full review of this issue is going to require some in depth analysis and telemetry for a snapshot of key IMU variables all taken at 200HZ, while the plane is flying.

phollands, 2016-12-20:
Stating this more clearly: The reason that IMUvelocity is more sensitive to location_error_earth, than IMUlocation, is that location_error_earth is synthetically applied as a velocity delta to IMUvelocity **immediately**  when location_error_earth is updated. It is an immediate update, and does not update slowly over a number of frames. (that is algorithmically correct, as the velocity has changed immediately to reflect the gradual change of estimated position as a result of location_error_earth).

So this is the reason that the location_error_earth related noise in IMU velocity is potentially around 5 times greater than the noise seen in IMUlocation.

phollands, 2017-01-07:
I have run some tests in HILSIM. 

I first created a base flight with air speed noise apparent. I took some of Bill's advice and

- [ took out GPS latency adjustment code for HILSIM,](https://github.com/phollands/MatrixPilot/commit/317d6a011c5a6ce569b53332fed28cf0b14a23c3) as there is no latency in the synthetic GPS in HILSIM, 

- [and also re-enabled normal airspeed calculations in HILSIM](https://github.com/phollands/MatrixPilot/commit/409e4e2bde436a8c29613d40f7c08d568e096a69)

 then I ran the code, but with telemetry being captured at 40 frames / second on a UDB4 instead of 4 frames / second.   I enclose a screenshot of a short flight (0003) up the runway at Innsbruck in X-plane 10, below.

[image]

I then took out some code in deadReckoning.c that previously [created a synthetic velocity adjustment based on errorLocationEarth. ](https://github.com/phollands/MatrixPilot/commit/873785c76b87c246173e4f9ce37e0a77012aa785)This pretty much removed the offending noise; see below (Flight 0005)....

[image]

You can see where I have commented out the relevant synthetic velocity code  below:- 

```
if (dead_reckon_clock > 0)
		// apply drift adjustments only while valid GPS data is in force.
		// This is done with a countdown clock that gets reset each time new data comes in.
		{
			dead_reckon_clock --;

			IMUintegralAccelerationx.WW += __builtin_mulss(DR_FILTER_GAIN, velocityErrorEarth[0]);
			IMUintegralAccelerationy.WW += __builtin_mulss(DR_FILTER_GAIN, velocityErrorEarth[1]);
			IMUintegralAccelerationz.WW += __builtin_mulss(DR_FILTER_GAIN, velocityErrorEarth[2]);

			IMUlocationx.WW += __builtin_mulss(DR_FILTER_GAIN, locationErrorEarth[0]);
			IMUlocationy.WW += __builtin_mulss(DR_FILTER_GAIN, locationErrorEarth[1]);
			IMUlocationz.WW += __builtin_mulss(DR_FILTER_GAIN, locationErrorEarth[2]);

			IMUvelocityx.WW = IMUintegralAccelerationx.WW;
			                  //+ __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[0]);
			IMUvelocityy.WW = IMUintegralAccelerationy.WW;
			                  //+ __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[1]);
			IMUvelocityz.WW = IMUintegralAccelerationz.WW;
			                  //+ __builtin_mulus(ONE_OVER_TAU, 100*locationErrorEarth[2]);

		}
```

phollands, 2017-01-19:
Bill has thought about the noise in the airspeed, and his recommendation to me is to put a filter on the calculated airspeed to cut out the noise. Kees, have you already done that ? If you have, can you point me to your code.

In the mean time, I have obstinately been trying to understand the source of the noise, which is seen upstream in  errorLocationEarth. I have done 27 flight in HILSIM with various experiments in place. You can see experiments in my [hilsim_easystar2 branch commits](https://github.com/phollands/MatrixPilot/commits/hilsim_easy_star2), and in the [imu_in_depth_telemetry](https://github.com/phollands/MatrixPilot/commits/imu_telemetry_in_depth)_commits.  

I enclose  a block diagram which I am using for methodically looking at aspects of Dead Reckoning.

Best wishes, Pete

[DeadReckoning.pdf](https://github.com/MatrixPilot/MatrixPilot/files/717851/DeadReckoning.pdf)

KeesGuijt, 2017-01-20:
Hi Pete,

The filter i wrote is [here](https://github.com/KeesGuijt/MatrixPilot/commit/2613eaa6477ed54a99fb48221000ed91362bb5e5).
The filter only updates the airspeed at 1Hz, using the latest calculated airspeed as 1 part, and 7 parts of the old airspeed.
As you know, and as reflected in your nice blockdiagram, the airspeed is not used for feedback in dead reckoning. 

Best wishes, Kees

phollands, 2017-01-20:
Thanks Kees for the[ link to your filter. ](https://github.com/KeesGuijt/MatrixPilot/commit/2613eaa6477ed54a99fb48221000ed91362bb5e5)

In the end my digging found a bug. There was a bug in a maths routine in mathlibNAV.c called "long_scale". The bug was causing GPSlocation.x (Meters East from Origin) to have a granularity of 4 meters. i.e. there could never be a GPSlocation.x of 3 7 5 9 meters, it always had to be a multiple of 4. e.g. 4,1,16. This was then injecting noise into locatoinErrorEarth[0] (Earth X axis). Fortunately all my tests had been with an East bound flight in HILSIM down the runway at Innsbruck. 

[I made a fix for long_scale() , and the Bill reviewed it and fixed a mistake that I made.](https://github.com/phollands/MatrixPilot/commit/ffdeb8c1892f8708cf34446d11245b6b84e1e66f) The end result is that the noise in locationErrorEarth[0]  is greatly reduced. My best run so far (with a best case setup in HILSIM, like having the magnetometer switched on, and arranging for no wind), is shown in the graph below. 

[image]

The associated graph of the airspeed for the same flight is shown below:-

[image]

So even with only a 1 meter error location, with TAU set to 1 second (rather than 2.5 originally), there will still be a 1m/sec noise introduced by errorLocationEarth to the air speed.  So, without filtering, one really would beed an errorLocationEarth that is in the tens of centimeters. That would involve a substantial re-working of the IMU in MatrixPIlot.  I'm not sure I'm ready for that. Is the effort really worth the return ?  For Quads hovering in one position, a centimeter accurate dead reckoning would be very desirable. But not for planes. 

Anyway, it's been nice to trace the main noise culprit to a bug. And I'll discuss with Bill about potential improvements to the IMU to reduce errorLocationEarth further.

The most likely outcome for now will be to introduce a filter on airspeed in master.

Best wishes, Pete

KeesGuijt, 2017-01-21:
Pete, 

Well done! So this may mean that my filter won't need to be as drastic anymore. Looks like it could be changed to 40Hz, not 8 Hz?

Kees

phollands, 2017-01-21:
Kees, 

Perhaps we should just discuss having an airspeed variable that does not include the synthetic airspeed component created from locationErrorEarth.;  ( airspeed_with_no_location_error ). All of the noise in the HILSIM test is attributable to locationErrorEarth. If we take that out of the calculation, then we have a completely dynamic airspeed with no filtering, which will also not suffer so much at higher speed from the GPS comms latency effects. A long GPS comms latency increases locationErrorEarth at high speeds, which in turn will increase the synthetic airs speed component associated with locationErrorEarth.

I'm going to have a chat with Bill about this soon, and I'll see what he thinks as well. If you would like me to send you over any of the raw test files, for you to analyse statistically,  then I'm happy to do so.

I hope to  re-run a real flight soon with the bug fix to long_scale() on a UDB5 to see how that has improved real flight. Also the UDB5, running at 200Hz DCM does reduce some of the GPS latency that is in the standard code logic, because it is running around it's cycle of updates so fast. My HILSIM tests and analysis to date are done on a UDB4 running at 40Hz.

Best wishes, Pete