MatrixPilot/MatrixPilot #108: SERIAL_UDB_EXTRA buffer indexing bug at high baud rates (e.g. 921600 ).

phollands, 2016-12-29:
While experimenting with Baud Rates of about 1 Mega Baud ( 921600 baud rate in options.h ) linking a UDB5 to a Raspberry Pi A+, I started to get repeating snippets and slightly garbled lines of F2: telemetry. 

Using a logic analyser I was able to show that I was not dropping characters but that the issue was occurring within the UDB5 and MatrixPilot. 

My current hypothesis is that while vsnprintf was readying and copying data to the serial buffer, the interrupt routine, which is now churning through it's characters at a high rate and at priority 4, and eating up the buffer, reaches the end of text \0 null character; but that the vsnprintf, running at priority 3, has by then overwritten the \0 null character marking the end of text, and is partway through copying it's new vsnprintf output to the buffer but has not yet finished. vsnprintf (priority 3) can be interrupted by the transmission part of the serial code (priority 4).  The result it that the firmware prints past the original end of text and finds old previously printed text to re-send until finding a second null char further down the buffer. I should stress that is a hypothesis and no absolutely proven. H

I have never noticed this issue before, but I think it arises because the interrupt routine is now so fast at 1 Megabaud, that it is reaching the end of line null character quicker, and while vsnprintf is still preparing  and copying over the next message.

I have already coded up one test solution that works by stopping the transmission of characters while vsnprintf is working and copying data into the serial buffer. That solution does fix the issue, and I'll post it as a Pull Request once I've thought about this some more, and done some more testing.

phollands, 2017-03-24:
Fixed in PR #135 .

closed 2017-03-24 by phollands

MatrixPilot, 2017-06-01:
another option that comes to mind would to the modify the uart tx interrupt
handler to replace each transmitted character in the buffer with a '\0'.