MatrixPilot/MatrixPilot #124: Barometers.c returns wrong pressure with Test Dataset for BMP180 / BMP085

phollands, 2017-02-17:
If I define TEST_WITH_DATASHEET_VALUES in barometer.c in master,
and compare the result in the telemetry with the expected [answer from the data sheet](https://cdn-shop.adafruit.com/datasheets/BST-BMP180-DS000-09.pdf) [See page 15], the resulting answer for the pressure is wrong. The relevant portion of telemetry from the F2: format line of SERIAL_UDB_EXTRA is:-

:tmp150:**prs7831**:alt0:

The temperature should show 150 (15 degrees), which it does.
The pressure should show 69964 but instead is showing 7831.

I had previously done a test flight using the barometer [Adafruit BMP180 ](https://www.adafruit.com/product/391) on a UDB5 providing the altitude. The altitude hold was good. But when viewed later on Google Earth, the Altitude of the GPS route when compared with the altitude of the IMU (based on the barometer) differed by more and more, as the plane flew higher and higher. So I then started to question whether the calculations in the barometer were correct. They were consistent, but they do not look correct.  

This is the first time I have used the barometer code in MatrixPilot.
I shall have a poke around and see if I can spot something obvious.

Best wishes, Pete

P.S. I enclose a screenshot of the KMZ GoogleEarth view showing the GPS track as much higher than the IMU track when the plane is high. But lower down, you will see that the two track re-converge and get closer together when the plane is close to the origin altitude at ground level.

[image]

elgarbe, 2017-02-17:
Peter, [here](https://github.com/MatrixPilot/MatrixPilot/blob/master/libUDB/barometer.c#L206) b7 is defined as LONG, but datasheet say that it need to be unsigned long. Could it be a source of error?

elgarbe, 2017-02-17:
Ac4, 5 and 6 need to be unsigned short and MP code define they as unsigned short (int for our 16 bit uC) [here](https://github.com/MatrixPilot/MatrixPilot/blob/master/libUDB/barometer.c#L43) so that is ok

KeesGuijt, 2017-02-17:
Pete, Leo,

The pressure value appears to be a bug. 
The difference in altitude between GPS and Barmeter is possible. The converging at lower altitudes could be a coincidence. Comparing GPS and Barometer really shows the strengths and weaknesses of both methods. I've found that (U-Blox) GPS drifts most in the first 10 minutes, after that + and - 20m is still possible. Barometer takes a few minutes to settle, then only drifts slowly with changing weather. I hope to do more testing on this also.

Regards,
Kees

phollands, 2017-02-17:
The reason that the Test Dataset was not working, was a mistake in the TEST Dataset code. Normally, the sensor is set to oversampling of "3" (which means oversample by 2^3  i.e. 8 times. But for running the test, the oversampling needs to be set to 0, and it was not being set to 0 by the code.  I enclose an example of a patch that fixes the test dataset. I will do a full pull request once I have found out why the pressure is not correct, for normal running of the code (not test running).

[image]

Leo, thanks for pointing out that error with the declaration of b7. I'll look at that carefully next.

phollands, 2017-02-17:
I have carefully added type casting to various parts of the code, and that seems to have fixed the pressure so that it is now reported correctly. The Sea Level Pressure calculated within MatrixPilot is now correct for my home location, when compared to pressure reported from the weather service.

I'll double check the tests tomorrow, and if all is still OK, I'll submit a pull request.

Best wishes, Pete

P.S. Leo the error that you caught (b7), did not fix the problem but I will include your fix in the Pull Request for good measure,

phollands, 2017-02-19:
Fixed in PR #125

closed 2017-02-19 by phollands