MatrixPilot/MatrixPilot #127: RSSI signal is not present on cvs file

elgarbe, 2017-02-22:
I would like to have RSSI signal to overlay in my dashware video. I don't have hurry on it.
Thank

phollands, 2017-02-23:
Acknowledged.

elgarbe, 2017-03-01:
Let me try to fix this issue, please. I study the changes you made to add barometer information into CSV and I think I could make similar changes to add RSSI.

phollands, 2017-03-01:
The barometer telemetry was already in SERIAL_UDB_EXTRA and in SUE over MAVLINK, so all I had to do was add the flan.pyw parser and add the fields the csv file production. But in the case of RSSI, the data is not currently in SERIAL_UDB_EXTRA. It can be easily added to  MAVLINK because that protocol is mature and already has fields for that.  The SUE over MAVLINK messages are now rather locked. A feature of MAVLINK is that it is unwise to change a given message, and it can then not be read by downstsream client programs (They use a CRC mechanism).  So I don't want to change SUE over MAVLINK, and that means I don't really want to change Ascii SUE anymore, because that would break backwards compatibility. I want to ensure that users that upgrade from SUE to MAVLink retain all the features of SUE. 

The easy way around all of this is to switch to MAVLINK which is a) more cpu efficient, b) more memory efficient c) has CRC redundancy checks built in.   Flan understands mavlink as well as SUE. So you will get all the SUE messages reconstituted into a LOGXXX.TXT file by flan.pyw. And Flan.py will still produce the csv file. So I think the correct way forward is to pick one of the 3 mavlink messages that incorporates RSSI, and add that into MatrixPilot. Then flan.pyw will need to be updated in it's parser (for mavlink this time) and the production of the csv file.

I hope that is clear. But let me know if you have any questions. The next step would be to discuss which of the 3 MAVLink messages containing RSSI should be used. 

You would need to try out MAVLink in your setup, as it is. It is meant to be fully backwards compatible, so if you capture the binary MAVLink file, and point flan.py at the file, then it should process it in the normal way. I know this all works with OpenLog, but I'm not sure which data logger you are using. It needs to be able accept a binary stream of data instead of Ascii.

phollands, 2017-03-01:
[Do you think that mavlink message](http://mavlink.org/messages/common#RADIO_STATUS) looks like a good candidate ?

phollands, 2017-03-01:
Actually I think that the above message is for the Telemetry Radio, not for the Control Radio. So may be we would need to use the RC_CHANNELS_RAW ( #35 ) message.

phollands, 2017-03-01:
Looks like someone has already [put rssi into our Mavlink implementation.](https://github.com/MatrixPilot/MatrixPilot/blob/master/MatrixPilot/MAVLink.c#L988l)

So if you can MAVlink, then, in theory ( :-) , it will just be a matter of updating flan.pyw to parse the mavlink message, and then produce the csv file again. I am happy to do that, if you could experiment with running MAVLink and check that SUE over MAVLink works for you, and produces the usual ascii csv file.

elgarbe, 2017-03-01:
I have my own Logger around a LPC11u67 (ARM cortex M0+) and it accept binary data... I think...
I will try SUE over mavlink and let you know the result. 
I have a couple off 900MHz radio link that could add to my setup to play a litle with MAVLink too.

Thank!