MatrixPilot/MatrixPilot #139: Hi. Couple of questions.

N-flyeR, 2017-05-18:
Hello. I have a AUAV (ArsovUAV), a UAV dev-board based autopilot version 1.1 (AUAV1).
I am looking for the all required files/firmware/flashing files/MatrixPilot required to flash my autopilot.
Could someone point me in the right direction and also tell me which is the latest MatrixPilot version that is compatible with AUAV1?
And yes, I did go to the Arsov website and I did talk to Nick Arsow himself and he redirected me to github. Unfortunately I am new to this website and it a bit hard to find where to download firmware files. If you could just tell me what I need to grab to get started I will be really grateful. 
And thank you guys for the awesome work!
(PS I also have some MatrixPilot quite interesting questions, related to setting up the autopilot but I will ask later once I got all needed files.)

phollands, 2017-05-21:
Sadly the Auav1 was created a long time ago. MatrixPilot today supports the
 AUAV3, the UDB4, the UDB5 and the UDB5mini.

Questions amd general discussuions should be made on the uavdevboard google
discussion group where you will get a much wider readership and response.

Best wishes Pete

N-flyeR, 2017-05-21:
Thank you sir for the kind reply. I did download all MatrixPilot versions on my PC. I now have only ONE single question: are you aware which is the latest MatrixPilot version that supports AUAV1? Thank you again for the awesome work.
And the google groups are pretty 'dead' for quite a long time, this is why I decided to ask you directly.

Inspirati, 2017-05-21:
> I think it would be reasonable to call the next release MatrixPilot 4.0,
> and remove support for older, pre-UDB4 boards.  And if we find bugs, or
> small improvements, we could backport them to the MP3.x brach and continue
> to make new releases of that branch too.
>
> But if it's easy to add features in a way such that they will only compile
> for UDB4, and UDB3 will still work without those features, then I think
> it would be better to keep supporting all the boards with one code base.
>
> So I guess I'm saying that if there's a major gain to be won by removing
> support for older boards, then I'm for it, otherwise not.
>
> Ben
>
>
>
> On Fri, Oct 19, 2012 at 3:22 PM, Robert Dickenson <robert.d...@gmail.com>

I believe 3.3 was the final version where we still worked on compatibility
with boards prior to UDB4.

https://github.com/MatrixPilot/MatrixPilot/tree/MatrixPilot_3_x

There was some brief discussion in the mailing list at the time. IIRC
William actually posted an official statement declaring the final
supporting version, however I can't find exactly that post just now.

Regards, Robert.

[UDB-dev] MatrixPilot 3.3 RC2
UDB_Dev
x
ben levitt <benjie@gmail.com> Unsubscribe
11/10/2012
to uavdevboard-dev
Hi guys,

I just uploaded MatrixPilot 3.3 RC2.  I don't know of anything else that we
still need to add or fix, so hopefully after a bit more testing, we can
turn this into an official release.
crashmatt uavflightdirector@gmail.com via
<https://support.google.com/mail/answer/1311182?hl=en-GB> googlegroups.com
Unsubscribe
11/10/2012
to uavdevboard-dev
Ben,

I should carefully clean out some of the advanced options that I don't wish
to document/support in this release. Having said this, I should not let it
delay the release.

Regards Matt
ben levitt <benjie@gmail.com> Unsubscribe
11/10/2012
to uavdevboard-dev
Hi Matt,

I merged the changes that you had made in your earlier b3.3 branch.  Do you
think that's good enough for now?
crashmatt uavflightdirector@gmail.com via
<https://support.google.com/mail/answer/1311182?hl=en-GB> googlegroups.com
Unsubscribe
11/10/2012
to uavdevboard-dev
That is good enough.  It had most of the junk removed.
William Premerlani <wpremerlani@gmail.com> Unsubscribe
11/10/2012
to uavdevboard-dev
sounds good, Ben.
Robert Dickenson <robert.dickenson@gmail.com>
20/10/2012
to uavdevboard-dev
Hi guys,

After this release, are we considering dropping support for the earlier
(pre UDB4) boards?

I am keen to move forward with merging my work in the UDB5 branch into
trunk, subject to a review and approval process.

Please let me know your thoughts.

Kind regards, Robert.
ben levitt <benjie@gmail.com> Unsubscribe
20/10/2012
to uavdevboard-dev
I think it would be reasonable to call the next release MatrixPilot 4.0,
and remove support for older, pre-UDB4 boards.  And if we find bugs, or
small improvements, we could backport them to the MP3.x brach and continue
to make new releases of that branch too.

But if it's easy to add features in a way such that they will only compile
for UDB4, and UDB3 will still work without those features, then I think it
would be better to keep supporting all the boards with one code base.

So I guess I'm saying that if there's a major gain to be won by removing
support for older boards, then I'm for it, otherwise not.
Mark Whitehorn kd0aij@gmail.com via
<https://support.google.com/mail/answer/1311182?hl=en-GB> googlegroups.com
Unsubscribe
20/10/2012
to uavdevboard-dev
Team,
I think there's merit to dropping support for pre-UDB4 boards in
MatrixPilot 4.0, since it would simplify the source by eliminating a lot of
preprocessor conditionals.

Additionally, new code could be written for UDB4 and later boards with all
board specific parameters localized to configBoardType.h instead of using
conditional compilation. My opinion is that it's better to have
parameterized algorithms with parameters specified as constants in
configXXX.h than to conditionally compile multiple code blocks. In Robert's
merged trunk/quad code, HEARTBEAT_HZ and PID_HZ are examples of this with
PID_HZ not yet fully implemented.

Since I, for one, do not have access to any pre-UDB4 boards, there won't be
any (pre-UDB4 tested) bugfixes coming from my direction anyway, and I
*might* be more productive with a simplified codebase. (and yes, I am
biased against conditional compilation)

73,
--Mark

closed 2017-05-30 by phollands