MatrixPilot/MatrixPilot #146: Custom Offsets don't allow digit 9

elgarbe, 2017-07-06:
I try to compile master with custom offset enable and get an error if some of the offset have a digit 9:

"C:\Program Files (x86)\Microchip\xc16\v1.26\bin\xc16-gcc.exe"   ../../libUDB/24LC256.c  -o build/default/production/_ext/1077454706/24LC256.o  -c -mcpu=33FJ256GP710A  -MMD -MF "build/default/production/_ext/1077454706/24LC256.o.d"        -g -omf=elf -DXPRJ_default=default  -legacy-libc    -mlarge-code -mlarge-data -O0 -I"../../Config" -I"../../MAVLink/include" -I"../../Microchip" -I"../../Microchip/Include" -I"../../libVectorMatrix" -DUDB5=1 -msmart-io=1 -Wall -msfr-warn=off  
../../libUDB/ADchannel.c: In function 'udb_a2d_record_offsets':
../../libUDB/ADchannel.c:79:22: error: invalid digit "9" in octal constant
../../libUDB/ADchannel.c:82:23: error: invalid digit "9" in octal constant
nbproject/Makefile-default.mk:1101: recipe for target 'build/default/production/_ext/1077454706/ADchannel.o' failed
make[2]: *** [build/default/production/_ext/1077454706/ADchannel.o] Error 255

elgarbe, 2017-07-06:
never mind, the problem is left 0's. If I use 009 got error, but no error using 9.
Sorry

phollands, 2017-09-04:
A leading 0 is seen by the compiler as denoting an Octal number. It is vital not to use leading zero's, or you may be defining a number in Octal instead of Decimal.

closed 2017-09-04 by phollands