MatrixPilot/MatrixPilot #172: Support STM

Linjieqiang, 2017-10-13:


phollands, 2017-10-13:
Not supported Today. Although Robert has done quite a bit of work on making the code portable to  ARM architectures. But it is not released and tested as far as I know.

 MatrixPilot is predominantly running on the Microchip dspic architectures.

Inspirati, 2017-10-13:
Thanks Pete, however my efforts were truly just as you suggest, aka,
towards making the build system somewhat compatible. Credit for the STM32F
series work belongs with LeoG, whom strode somewhat forwards toward making
it work (aka, he sent me his work, in terms of hardware home made adapter
boards!!), alas but alas, nobody else here followed....

Linjieqiang, 2017-10-13:
That's a big pity.Thanks.Hope that day coming.

closed 2017-10-14 by phollands